Aharoni-Lab/Miniscope-v4 #21: Atmega328‘s  Downloader

931436690, 2020-07-08:
Thank you for the contribution of the system. We are studying it.
But we have one question now. What downloader are using for the atmega328? We can't find  1.8V downloader for the atmega328. Most of atmega328‘s  downloader in the market is 5V with USB supplying the power.We think 1.8V is the best voltage for Python480.
Best wishes!

daharoni, 2020-07-08:
I think almost any ISP Programmer would work for this. ISP has a Voltage Target (VTG) pin which the programmer uses to know what voltage to communicate at.

We use the AVR Dragon or Atmel Ice usually.

closed 2020-07-08 by daharoni

931436690, 2020-07-08:
Thank you very much. We think  Atmel Ice is the option for us. 
Best wishes.