Aharoni-Lab/Miniscope-v4 #25: The thin ring of EWL, changing PCB

peng831, 2021-02-20:
Hello there,
  We want to enlarge the thin ring of EWL, the original ring is difficult to assemble. Do you think any problems will occur? I am worried that this change will adversely affect the camera.
The change looks like this：
[image]

Sorry to ask this question repeatedly， I cancelled it for the first time.

fnsangiul, 2021-02-20:
Unfortunately that bigger diameter is not going to work. Not only because it's not going to fit in the objective module, it also won't properly make contact with the EWL.

peng831, 2021-02-21:
Why can't it contact EWL correctly? In my opinion, it is a wire connecting the liquid lens, just like the thick ring on the other side of the liquid lens (about 282°  ring). What we want to do is to replace 'the original thin ring' to 'the bigger ring',which is like the right side ring of the picture.

peng831, 2021-02-21:
We have measured the size of the liquid lens; if a larger ring is used, the liquid lens can touch a larger area.

fnsangiul, 2021-02-21:
Electrically there shouldn't be any problem, but mechanically the bigger ring won't fit in the objective module and when pushed inside it won't make contact. We had a similar issue while validating the design of the objective module.

peng831, 2021-02-21:
We are more worried about the impact of electrical performance; we can slightly modify the objective lens module to adapt to the large ring, in order to better assemble the liquid lens. The radius of the red ring is 3.72mm, and the line width is 0.6mm; the radii of the two yellow rings are 3.3mm and 4.3mm respectively.

fnsangiul, 2021-02-21:
I think that the easiest solution for you is to test your new design with the pads we left on the bottom on the PCB

[image]

[PCBway](https://www.pcbway.com/) has the capabilities to fabricate them. I hope this helps

peng831, 2021-02-22:
It's a good idea. Thank you for your help.

closed 2021-03-03 by daharoni