Aharoni-Lab/Miniscope-v4 #32: KiCad file Miniscope-v4-Rigid-Flex-cache.lib is missing

alexanderwiller, 2023-01-11:
**Describe the bug**
The KiCad schematic cannot be loaded properly, as the cached library `Miniscope-v4-Rigid-Flex-cache.lib`, which contains some of the symbols used, is missing.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the repository.
2. Open the project with KiCad
3. Open the schematic

**Expected behavior**
The library is included in the repository. The schematic can be loaded properly.

**Screenshots**
[image]
[image]

**Additional context**
We would like to convert the Miniscope V4 PCB to KiCad V6, add an automation pipeline for production data export with KiBot and make some small improvements, e.g. add radii where the traces meet the EWL contact rings, to increase robustness. This is closely related to the same effort that we go to with the Miniscope-LFOV: https://github.com/Aharoni-Lab/Miniscope-LFOV/issues/3

jonnew, 2023-01-11:
This was a major issue with older version of KiCAD. You will need to add the Aharoni Lab KiCAD lib (https://github.com/Aharoni-Lab/KiCad-Library) to KiCAD and format all your library paths to match wherever its trying to look for those missing symbols. Once you do this and then upgrade to V6 (or V7, RC candidate is now available). The symbols will be embedded in the schematic and this will no longer be an issue.

alexanderwiller, 2023-01-11:
Thanks for the reply, unfortunately, the link does not work for me. Is this a public repository?

I count on the embedding feature, it is very useful for distribution of projects. This already worked well for the Miniscope-LFOV 👍 For that project, the rescue lib was present and I could delete it after the conversion.

KiCad 7 RC, I prefer to not use yet, as I don't know how well it will work with KiBot etc.

jonnew, 2023-01-11:
Huh, it is private. I don't know why I'm a member in that case since I'm not in the lab. @daharoni why is the KiCAD lib private?

daharoni, 2023-01-11:
Hi! I just made our Kicad Library public. We will also try to port our projects over to Kicad 6 along with better embedded library management when we have time.

closed 2023-01-11 by daharoni

Ildaron, 2025-01-15:
Hi @jonnew  can you say help how to transfer  Aharoni Lab KiCAD lib (https://github.com/Aharoni-Lab/KiCad-Library) to my new kicad ? I indicated way

[image]

Ildaron, 2025-01-17:
Fixed , just updated kicad to 8.0.8