Aharoni-Lab/Miniscope-v4 #37: miniscope brightness <1?

hsw28, 2024-01-30:
would love if the @miniscopeteam brightness could go below 1. I tried setting it to 0.5 in the config but unsure it actually worked. (yes gain is already down... I could increase sampling rate then downsample later but that is annoying to have to do for one animal only :-p)

@sneakers-the-rat did a nice writeup on this as well here: https://github.com/Aharoni-Lab/Miniscope-DAQ-QT-Software/issues/68

fnsangiul, 2024-01-30:
Hello @hsw28. To achieve brightness levels lower than 1 (on the Miniscope Software scale) you will need to do a slight modification to the hardware. On the excitation PCB you will need to replace the R1 resistor (shown below in the red rectangle)

[image]

On the v4 Miniscope this resistor has a value of 10.2 kΩ. To achieve lower levels of brightness you have to use a 100 kΩ resistor. [This resistor](https://www.digikey.com/en/products/detail/panasonic-electronic-components/ERJ-1GNF1003C/2036340) can be an option. 

Please let us know how if you have any questions.

hsw28, 2024-01-30:
Ok so it's not possible to do this on the software side? bc i would also like to use the same scope for the dim guys :)

fnsangiul, 2024-01-30:
Another option is to use a userConfig file that lowers the exposure of the Miniscope. Then you wouldn't need to do this hardware modification and you can switch to the normal configuration for the dim mice. I'll this userConfig later.

hsw28, 2024-01-30:
that would be superb!! thank you!

sneakers-the-rat, 2024-01-31:
Do we know what the commonly used brightness values are? If its most often the case that ppl want the LEDs dimmer, we might consider updating the design. Obvi that will be a big range, dependent on the indicator, expression, imaging condition, etc., so what im asking is if we have any kinda sense of that range - do ppl use all of it, or does everyone stay around 0-10 or whatever, because youre right it would be awesome to decrease bleaching and give ppl more usable dynamic range

fnsangiul, 2024-01-31:
@hsw28 Assuming you are using the latest version of the software (v1.11) please replace the attached files "videoDevices" and "miniscopes" on the "deviceConfigs" folder. Also use the attached user configuration file ("UserConfigExample-Singlev4_Exposure_limit"). This modifications will give you 30 FPS with exposure ranging from 100% to 20% in 10% increments.

[UserConfigExample-Singlev4_Exposure_limit.json](https://github.com/Aharoni-Lab/Miniscope-v4/files/14106252/UserConfigExample-Singlev4_Exposure_limit.json)
[miniscopes.json](https://github.com/Aharoni-Lab/Miniscope-v4/files/14106253/miniscopes.json)
[videoDevices.json](https://github.com/Aharoni-Lab/Miniscope-v4/files/14106254/videoDevices.json)

Hope this helps and don't hesitate to ask any questions.

sneakers-the-rat, 2024-01-31:
Should we add that to the DAQ software repo and add a docs markdown page to describe what it does?

hsw28, 2024-01-31:
@fnsangiul thank you!! the default exposure is 100 I assume?

hsw28, 2024-01-31:
> @hsw28 Assuming you are using the latest version of the software (v1.11) please replace the attached files "videoDevices" and "miniscopes" on the "deviceConfigs" folder. Also use the attached user [...]

Sorry, I cant actually find in these files where I set the exposure value

hsw28, 2024-02-01:
@fnsangiul -- Sorry, I cant actually find in these files where I set the exposure value

fnsangiul, 2024-02-01:
@hsw28 I don't have a Miniscope with me right now so I can't attach screenshots. The reduced exposure values can be found on the 30 FPS option. As you continue clicking the exposure will reduce by 10%. I'll post screenshots on this tomorrow.

sneakers-the-rat, 2024-02-02:
fedi def knows this fix better than me, but a diff of the `videoDevices.json` file from the regular one looks like:

```diff
--- Miniscope-DAQ-QT-Software/deviceConfigs/videoDevices.json
+++ Miniscope-DAQ-QT-Software/deviceConfigs/videoDevices_exposure.json
@@ -258,7 +258,7 @@
 
         ]
     },
-    "Miniscope_V4_BNO": {
+    "Miniscope_V4_BNO_Exposure_Limit": {
         "qmlFile": "qrc:/Miniscope_V4_BNO.qml",
         "sensor": "PYTHON480",
         "frameRate": "adjustable",
@@ -288,9 +288,9 @@
                 ]
             },
             "frameRate": {
-                "displaySpinBoxValues":["2FPS","5FPS","10FPS", "15FPS", "20FPS", "25FPS", "30FPS"],
-                "displayTextValues": [2, 5, 10, 15, 20, 25, 30],
-                "outputValues":[50000,20000,10000,6667,5000, 4000, 3300],
+                "displaySpinBoxValues":["2FPS","5FPS","10FPS", "15FPS", "20FPS", "25FPS", "30FPS", "30FPS/90", "30FPS/80", "30FPS/70", "30FPS/60", "30FPS/50", "30FPS/40", "30FPS/30", "30FPS/20"],
+                "displayTextValues": [2, 5, 10, 15, 20, 25, 30, 309, 308, 307, 306, 305, 304, 303, 302],
+                "outputValues":[50000,20000,10000,6667,5000, 4000, 3300, 2970, 2640, 2310, 1980, 1650, 1320, 990, 660],
                 "startValue": "20FPS",
                 "sendCommand": [
```

and the main change in the `UserConfig` example looks like this:

```diff
@@ -100,15 +103,15 @@
     "devices": {
         "miniscopes": {
             "miniscopeDeviceName": {
                 "COMMENT_deviceName": "This can be whatever you want. It should be unique for each device though.",
                 "COMMENT_deviceType": "This should be a supported deviceType listed in the videoDevice.json file.",
                 "COMMENT_deviceType2": "Supported types include 'Miniscope_V4_BNO' and 'Miniscope_V3'",
-                "deviceType": "Miniscope_V4_BNO",
+                "deviceType": "Miniscope_V4_BNO_Exposure_Limit",
                 "COMMENT_headOrientation": "If using a V4 Miniscope you should set enabled to true so stream and record head orientation data from the onboard IMU.",
 				"headOrientation": {
 					"enabled": true,
```

So it looks like you're looking for some new values in the framerate spinbox?